maniacbug/mighty-1284p #14: Arduino IDE 1.6.1

zevero, 2015-02-24:
Structure changed somehow in the current nigthly builds.
In order to work I had to add some files and parameters.

My fork is only for my Atmega1284p 16Mhz using avrispmkii

https://github.com/zevero/mighty-1284p/commit/341b383ee962237d3a56720255acfa2f0afdac44

Its quite strange that I had to state the uploader in the board file ...

zevero, 2015-04-30:
current nightly build was 1.6.1
It runs on 1.6.1 but not on 1.6.2 or 1.6.3
Board simply does not show up.
If I go to board manager, there are some complaints about deprecated paths. But when I do like advised, the complaints dissappear, but still: No Boards to choose from!

I am stuck.
I tried this other version of mighty1284p
https://github.com/wijnen/mighty-1284p
which does not mess in the arduino code but only uses the hardware folder in your sketch folder. But it is completely ignored in all Versions 1.6.x

per1234, 2015-04-30:
maniacbug's mighty 1284P has not been in development for years, there have been changes in the Arduino IDE since then that make it obsolete. The active fork is https://github.com/JChristensen/mighty-1284p but this still doesn't support Arduino IDE 1.6.x. I have updated it to work with 1.6.3 in my fork here: https://github.com/per1234/mighty-1284p If you use it, please let me know if you find any issues. Thanks, Per

zevero, 2015-04-30:
Thank y1ou!
It works.
Using AVRISP mkii was possible after going to Boards > Boardmanager ... and making an update.
It loads quite some stuff into .arduino15 and seems to use its own Version of avrdude!!!

While all works great know, I have lost understanding, what is going on. For example I would like to use avrdude -V in order to avoid waiting for an entire read after flashing, but changing programmers.txt or platforms.txt do not seem to have any effect.

The Bootloader installs fine now, but normal serial flashing does not work since it seems to try using avrdude nevertheless, complaining "not in sync" ...

Thank you nevertheless!

zevero, 2015-04-30:
Ok, i found that in
.arduino15/packages/arduino/hardware/avr/1.6.6/programmers.txt
I can set -V in front of -Pusb as an extra parameter

But I still dont understand the purpose of .arduino15 and why I should mess around there...

per1234, 2015-04-30:
I think you can get the same effect as avrdude -V by File>Preferences>Verify code after upload(uncheck). 

As for modifying platforms.txt I have not created a separate one for mighty1284P so it will use the one in your arduino15 folder unless you create a separate platforms.txt and place it in sketchbook\hardware\mighty-1284p\avr folder. Note that in IDE 15+ the folder structure is different than 1.0.x so there are 2 copies of all the files, one copy for each IDE version.

Serial flashing works for me in 1.6.3 with my fork but I had intermittent problems uploading larger files to my homemade 1284P 16Mhz dev board using the FTDI FT232RL. The solution was to create a new entry in boards.txt that sets the low fuse to 0xF7 for full swing oscillator:
```
mighty_opt_FS.name=Mighty 1284p 16MHz full swing using Optiboot
mighty_opt_FS.upload.tool=arduino:avrdude
mighty_opt_FS.upload.maximum_data_size=16384
mighty_opt_FS.upload.protocol=arduino
mighty_opt_FS.upload.maximum_size=130048
mighty_opt_FS.upload.speed=115200
mighty_opt_FS.bootloader.tool=arduino:avrdude
mighty_opt_FS.bootloader.low_fuses=0xf7
mighty_opt_FS.bootloader.high_fuses=0xde
mighty_opt_FS.bootloader.extended_fuses=0xfd
mighty_opt_FS.bootloader.file=optiboot/optiboot_atmega1284p.hex
mighty_opt_FS.bootloader.unlock_bits=0x3F
mighty_opt_FS.bootloader.lock_bits=0x0F
mighty_opt_FS.build.mcu=atmega1284p
mighty_opt_FS.build.f_cpu=16000000L
mighty_opt_FS.build.core=arduino:arduino
mighty_opt_FS.build.variant=standard
```

per1234, 2015-04-30:
The purpose of arduino15 is to allow different users to have different configurations instead of the old system where the files were located in the Arduino install folder. These are the standard configuration files. If you modify the files in the 3rd party hardware folder in your sketchbook folder then you are modifying the configuration for that specific hardware set. It is explained further here: https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5---3rd-party-Hardware-specification

zevero, 2015-04-30:
So there is the arduino-1.6.3 installation folder, the Arduino sketch folder and .arduino15 ... thats quite a lot! .arduino15 seems to replace large parts of the installation folder...

Upload via FTDI is after many attempts still not possible.

         Using Port                    : /dev/ttyUSB0
         Using Programmer              : arduino
         Overriding Baud Rate          : 115200
    avrdude: stk500_recv(): programmer is not responding
    avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x00

which is weired, because it should just talk via /dev/USB and not trying to use the mkii.

Uncheck of verify code does not stop avrdude frome verifying.

But I am happy with my own platforms.txt in sketchbook\hardware\mighty-1284p\avr
and everything is running great!

Thanks. After all of the code I have seen today, after so many people not having been able to port migthy1284p to 1.6.x, you truly are a hero!

per1234, 2015-04-30:
When you Upload Using Programmer the bootloader is erased so you have to Burn Bootloader again after Upload Using Programmer if you want to flash via serial. If you try to flash using serial you will get the "not in sync" error so give that a try.

I'm very glad to hear my work has been useful to you, that makes the extra effort of making it publicly available worth it. It's nice to be able to make a small contribution to the great work of maniacbug and JChristensen. I hope it will be merged soon, I was hoping to find someone to test my update of the patched GSM library before making the pull request but haven't had any luck.

per1234, 2015-05-07:
Mighty 1284P is now Arduino IDE 1.6.x compatible: https://github.com/JChristensen/mighty-1284p/tree/v1.6.3

zevero, 2015-05-08:
GREAT!
I testet it and removed old 1.6.3 installation and .arduino and .arduino15 directory and Arduino/hardware  dirs and started from scratch on my Debian Linux.
Downloaded arduino-1.6.3, downloaded this Version of mighty and ...
:+1:  It works right out of the box!

The avrispmkII programmer did not work (avrdude not compiled for device ... ). But pressing Update once in Boardmanager loads 30 Mb and then it works :+1:

ShriHari007175, 2016-06-19:
Dear Sir,

I have Atmega1284P rectangular or Squared Microcontroller and I have marlin firmware software. How do I bootloader marlin code into 1284P. Please tell me the steps. It would take only few minutes for an experts but for me it may be months and months. I am beginner for this field. I am learning. I have seen some tutorials and also went for some web sites, blogspots and all. I did not get which PROGRAMMER from arduino we should select. Even I see How arduino UNO can be act as Arduino by ARDUNO-ISP for Atmega328. I did not get steps or videos related to ATmega1284P. So, kindly do a favor in telling me that HOW TO BOOTLOAD Atmega1284P with what PROGRAMMER? and what all hardwares needed for it like PROGRAMMER device and all. I needed step by step procedure. Please take your valuable time.

Regards,

shri.7175@gmail.com

ShriHari