maniacbug/mighty-1284p #17: problem with the MCUSR

jar10623, 2015-11-29:
on a arduino nano328p this code worked fine:

before setup:
  uint8_t my_mcusr=MCUSR;
in loop:
  myGLCD.print("my_mcusr= ", LEFT, LINE7);
  sprintf(r_str, "\"%03d\"", my_mcusr);
  myGLCD.print(r_str, RIGHT, LINE7);

connected with PC in the Arduino IDE
on nano328p output MCUSR dec 6
on mighty1284p output MCUSR dec 0

on power supply with powerON
on nano328p output MCUSR dec 5
on mighty1284p output MCUSR dec 0

on power supply with push resetSWITCH
on nano328p output MCUSR dec 7
on mighty1284p output MCUSR dec 0

has anyone an idea?

baldowl, 2015-11-29:
It should be the bootloader: assuming you're using the old optiboot version bundled with this code, it resets MCUSR before starting your code.

jar10623, 2015-11-29:
is there any correction?

maybe a new bootloader.hex for 1M Baud Optiboot and 16MHz F_CPU?

baldowl, 2015-11-29:
I think it's better if you ask in the more active fork: https://github.com/JChristensen/mighty-1284p

Or you can compile and replace the bootloader: https://github.com/Optiboot/optiboot

jar10623, 2015-11-29:
thanks, soory i don't can compile it by myself and the Link only gives
a HEX file to

atmega1284o.name=[Optiboot] ATmega1284p
atmega1284o.upload.speed=115200

and not to 1M upload

baldowl, 2015-11-30:
Ok, then have a look at https://gist.github.com/baldowl/e6593d53f2454159571b

I compiled the Optiboot's current master using the tool chain bundled in Arduino's IDE version 1.6.6, but, mind you, it doesn't matter so you don't have to upgrade IDE or whatnot: just burn the bootloader.

Note that MCUSR is still reset but its value is saved in R2 so it's available for your application.