maniacbug/mighty-1284p #18: Can't get bootloader loaded, ever.

Davewerld, 2016-03-17:
I have followed in detail Maniacbugs forum posting for developing on the ATMEGA1284, I checked all my hardware and made sure the oscillator was working, I checked all voltages. The hardware seems correct. I tried using the Arduino as ISP and although the tx/rx lights flicker, it says it burned a bootloader, but it hasn't. I can't see blinking on LED on pin 1 upon power cycle. I also can't upload a program to it using a working FTDI programmer. So I tried Gammon's Atmega_Board_Programmer and although it claimed success, I see no LED blink and cannot download to the board either. So I abandoned Arduino 1.0.1 and moved to 1.6.8 and followed the instructions here. Still no success. The latest issue is when using ArduinoISP, avrdude says avrdude: Expected signature for ATmega1284P is 1E 97 05
         Double check chip, or use -F to override this check.
Wrong microcontroller found.  Did you select the right board from the Tools > Board menu?

Mr Gammon's board programmer tells me the signature is in fact 0x1E, 0x97, 0x05, so I don't understand that error. what I do understand is that after about 9 hours I still haven't downloaded a bootloader and I don't understand why. I have tried it on an ATMEGA328 and it worked like a champ, so I don't think I am doing much wrong there. I also have ripped up and redid the breadboard twice following Maniacbug's schematic to eliminate that I made a pin error. My ATMEGA is a ATMEGA1284P-PU. I am using a 16MHz resonator and an Arduino UNO as the programmer. Does anyone have any idea why this isn't working at all?

Agent24, 2016-03-17:
You should probably post this on the forum rather than as a bug report because it's probably some mistake rather than a bug. I know when I was first playing around with these it took a long time to figure it out as well.

Have you added the correct details for the 1284p to your boards.txt list, and added the files to the cores\variants folders? You didn't mention these, so I'm asking just in case.
Have you tried to manually program the bootloader file with avrdude through the commandline?
If you bought a brand new AVR, did you program the fuses correctly?

Davewerld, 2016-03-17:
Sorry, I didn't know where to post it, I'm not too familiar. I have not added the correct details for the 1284p to my boards.txt list, or added files to cores/variants. I followed the instructions for unzipping and placing all the info in the Arduino hardware folder. It shows up on the IDE as a valid board, so I had no idea I need to put anything else anywhere. It is a brand new AVR, isn't loading a bootloader how I program it to accept an arduino program. I don't know what fuses to load. 

I haven't used avrdude directly either, I didn't want to add variables of things I don't know, which explaining back to you, looks like a lot.

Agent24, 2016-03-17:
Honestly I am not sure if this is a bug or incompatibility or what. I don't know why it didn't work with 1.0.1 but for 1.6.x it shouldn't work at all: https://github.com/maniacbug/mighty-1284p/issues/14

They say you should try the version here: https://github.com/JChristensen/mighty-1284p/tree/v1.6.3

If you unpacked all the correct files it should have added them for you so I don't think that's the problem.
Yes, burning the bootloader should set the fuses for you, sorry, I forgot about that part. It's been a while.

If that updated fork of this project doesn't work still, I can try this myself as I have some 1284Ps around somewhere...

Davewerld, 2016-03-17:
I did move to the JChristensen /mighty-1284 when I migrated to 1.6.8.  I had been struggling with 1.0.1 and finally found this thread on the web. I'm convinced I am not burning a bootloader, even with the Gammon file.  Tomorrow's a new day, I'm hoping to solve this then. Thanks.

FedorAdarichev, 2018-01-07:
Hey Everybody,

I'm also having an issue with burning a bootloader.  I have a 1284P and I am using the guide for burning a bootloader found here: https://github.com/MCUdude/MightyCore.  I am running Arduino IDE v1.6.11.  I'm getting the following error:

> avrdude: Version 6.0.1, compiled on Apr 15 2015 at 19:59:58
         Copyright (c) 2000-2005 Brian Dean, http://www.bdmicro.com/
         Copyright (c) 2007-2009 Joerg Wunsch

         System wide configuration file is "C:\Users\ME\AppData\Local\Arduino15\packages\MightyCore\hardware\avr\1.0.6/avrdude.conf"

         Using Port                    : COM11
         Using Programmer              : stk500v1
         Overriding Baud Rate          : 19200
avrdude: stk500_getsync() attempt 1 of 10: not in sync: resp=0x60
avrdude: stk500_getsync() attempt 2 of 10: not in sync: resp=0x18
avrdude: stk500_getsync() attempt 3 of 10: not in sync: resp=0x18
avrdude: stk500_getsync() attempt 4 of 10: not in sync: resp=0x98
avrdude: stk500_getsync() attempt 5 of 10: not in sync: resp=0x1e
avrdude: stk500_getsync() attempt 6 of 10: not in sync: resp=0x98
avrdude: stk500_getsync() attempt 7 of 10: not in sync: resp=0x80
avrdude: stk500_getsync() attempt 8 of 10: not in sync: resp=0x60
avrdude: stk500_getsync() attempt 9 of 10: not in sync: resp=0x18
avrdude: stk500_getsync() attempt 10 of 10: not in sync: resp=0x18

avrdude done.  Thank you.

Error while burning bootloader.

Here is a picture of my hook-up:
[image]

I have a green wire soldered to the leg of my reset switch because I realized that the FTDI header port I have will have it going through a 0.1uF resistor, and not directly to the reset pin of the 1284P, so I Macgyvered it for the time being.  Also, I lost a 22pF SMT capacitor for one of the 16MHz oscillator capacitors, so I soldered a THT one to it instead. Besides that, I'm sure I have the layout of the PCB correct.

The settings I have when burning are:
[image]

Any ideas on what my issue could be?

Thanks,

per1234, 2018-01-07:
@FedorAdarichev post on the [Arduino forum](http://forum.arduino.cc/). I'm sure we will be able to help you over there. It's not appropriate or effective to ask for support here.

FedorAdarichev, 2018-01-07:
OK, thank you.