maniacbug/mighty-1284p #22: ERROR COMPILATION...

farmaceuta777, 2021-02-19:
hello ... I have a problem like in the picture..after adding the "hardware" folder and placing the file I have additional board options but I cannot compile the code ... why?
I'm sorry I'm a beginner ..: - /
(I don't have atmega1284 at the moment .. I'm just trying to check the code compilation "dry")

Best wishes..:-)
[image]

per1234, 2021-02-19:
@farmaceuta777 this project has been abandoned for almost a decade now. It is not compatible with modern Arduino IDE versions. Please use MightyCore:
https://github.com/MCUdude/MightyCore

farmaceuta777, 2021-02-19:
Hmmm ... I tried in the first place the version you gave, but I did not get a new board in the tools at all ..