maniacbug/mighty-1284p #23: Error while burning bootloader

ElectronsFellOut, 2021-09-14:
Hi, I have been struggling to program an ATMEGA1284P for quite some time now. I'm using an external 16MHZ crystal with relevant 22pf capacitors. I am trying to use an arduino nano as ISP using the arduinoISP sketch however when I select Mighty 1284p 16mhz using optiboot then try and burn bootloader I get this:

java.lang.NullPointerException
	at cc.arduino.packages.uploaders.SerialUploader.burnBootloader(SerialUploader.java:329)
	at processing.app.Editor.lambda$handleBurnBootloader$61(Editor.java:2379)
	at java.lang.Thread.run(Thread.java:748)
Error while burning bootloader.

Prior to this, I would just get programmer not in sync / responding errors while trying to upload sketches.
ATMEGA1284P is factory fresh.

Agent24, 2021-09-14:
I've never had much luck with the bootloader upload in Arduino. Usually resort to doing it manually with avrdude and a custom-built parallel-port interface circuit.

per1234, 2021-09-14:
Maintenance of this project ceased almost a decade ago. It has archaeological value for those who wish to study the pioneering work in 3rd party Arduino boards platforms, but not so much for actual use. I recommend using the actively maintained MightyCore:
https://github.com/MCUdude/MightyCore
And I second **Agent24** that you should ask for assistance on the forum if you have any problems with burning the bootloader using MightyCore.

ElectronsFellOut, 2021-09-22:
> Maintenance of this project ceased almost a decade ago. It has archaeological value for those who wish to study the pioneering work in 3rd party Arduino boards platforms, but not so much for actual [...]

Thanks for your reply, I got a USBasp programmer, used mightycore and it works!